JGAguado/Smart_Powermeter #8: No voltage measurement

agkaminski, 2023-07-25:
Please take a note that you cannot accurately measure AC power by only measuring current. Without information about phase shift between the voltage and the current you cannot measure real power usage - you can only measure apparent power, which is kinda useless.

Wikipedia has it laid out accurately enough: https://en.wikipedia.org/wiki/AC_power

JGAguado, 2023-07-25:
The current board provides a real-time **estimation**. Voltage measurement is a feature that will come in future versions.

closed 2023-07-25 by JGAguado

agkaminski, 2023-07-25:
You seem to misunderstand the issue - amplitude of the voltage is not really needed (as we can assume 230V and live with it), but the phase is necessary

JGAguado, 2023-07-25:
Dear Aleksander, I understand perfectly the issue. If you dig a little bit into the firmware my board is compatible with (ESPHome), you will find that the power factor computation is not supported a the moment. Since I don't take part in the firmware development, it made no sense for me to implement (at least in the first versions) the voltage measurement for computing the phase offset, true and reactive power.

Saentist, 2025-03-31:
@JGAguado 
Voltage can be taken from Home assistant
https://esphome.io/components/sensor/homeassistant.html

That's how, if any device report it already can be used.
Documentation and config file update needed also.