JGAguado/Smart_Powermeter #10: BOM and CPL Files Available?

mdframe, 2024-01-08:
Does anyone have all the files necessary to order the PCB from JLCPCP for board and components?  

Thank you, 

Matt

mdframe, 2024-01-08:
Is there any way to get notified when the next batch is ready for an order?  I appreciate the quick response and yes all I am looking to do is place an order. 

Thank you, 

Matt

JGAguado, 2024-01-08:
Let me come back to you on this channel when it becomes available.

JGAguado, 2024-01-09:
Dear @mdframe,

You are very welcome! Would be great to have you supporting the Smart Powermeter's features on ESPHome! And yes, please keep me posted on any updates you make!

At the moment, the published code, is focused on the measurement, and a "Hello World" test for the display. So if you would make any advance with some cool graphs or so, please keep me posted 😃 !!

The problem with the code running behind the Smart Powermeter shown in the photos, is that it required a Python script running in Home Assistant to convert the gathered power data from ESPHome into a scaled array (x,y,w,h), that I later passed back to the ESPHome to be plotted. Unfortunately, on one of the last HA updates, it stopped working and I need to find some time to fix it... Since it broke, I am just plotting the measured data with the standard points, which is not that nice, but does the job...

I will close this issue now, but for further questions don't hesitate to write me by email (jon.garcia.aguado@gmail.com) or IG (@JGAguado), I may answer faster.

Best regards and thanks for the support!

Jon

closed 2024-01-09 by JGAguado