JGAguado/Smart_Powermeter #13: Documentation - Assembly - Typo

mdframe, 2024-01-15:
Jon, 

I found a typo in the documentation detailing the ESPHome folder structure:

[image]

I am working through your code to verify what is working from a user stand point.  I am also having a really difficult time finding the correct webfont file and believe this is causing me errors with the  following line:

 it.printf(x_head, y_head, id(font_icon), TextAlign::TOP_LEFT, wifi_icon_map[4].c_str());

ESPHome compiler is stating no wifi_icon_map exists, here is the error:

<unicode string>: In lambda function:
<unicode string>:269:71: error: 'wifi_icon_map' was not declared in this scope
<unicode string>:269:71: note: suggested alternative: 'wifisignal'
<unicode string>:272:71: error: 'wifi_icon_map' was not declared in this scope
<unicode string>:272:71: note: suggested alternative: 'wifisignal'
<unicode string>:275:71: error: 'wifi_icon_map' was not declared in this scope
<unicode string>:275:71: note: suggested alternative: 'wifisignal'
<unicode string>:278:71: error: 'wifi_icon_map' was not declared in this scope
<unicode string>:278:71: note: suggested alternative: 'wifisignal'
<unicode string>:281:71: error: 'wifi_icon_map' was not declared in this scope
<unicode string>:281:71: note: suggested alternative: 'wifisignal'

Again, I think this is related to the ttf file I am trying to use since I can't find the defined one in the code.  This is the only item I have an error on at this point.   Just working through the YAML prior to receiving the hardware.  I will make sure this works locally before making recommendations or changing your code as it sits right now.

Thank you, 

Matt

JGAguado, 2024-01-15:
Matt,

Thanks a lot for opening the issue, it makes me really happy that at least someone read the docs hahaha. The tree typo has been corrected and I added a downloadable MDI .ttf like the one I use.

Can I close the issue or there is something else not compiling? 

Best regards!

Jon

mdframe, 2024-01-15:
Thanks Jon, I'm good at this point until I get hardware in hand!  ;)