JGAguado/Smart_Powermeter #14: Incorrect name of map in icon_map.h provided in documetnation

Brockhold, 2024-01-24:
I saw this was mentioned in #13 as well, but that issue is closed already. 
On line 2 `std::map<int, std::string> battery_icon_map` the name of the map should be `wifi_icon_map` instead.

JGAguado, 2024-01-24:
Dear @Brockhold,

Issue fixed and as _bonus_, I updated the MDI "fonts" to avoid needing local ttf files. Havent tested it yet, but if you do and nothing breaks, feel free to close the topic 😄 !

Best regards,

Jon

JGAguado, 2024-01-28:
Tested working, I will close the issue now. 

Btw @Brockhold, I created a discussion section on this repo, feel free to use it for sharing your Smart Plants applications, ESPHome examples, or any other topic 😉

closed 2024-01-28 by JGAguado