JGAguado/Smart_Powermeter #15: New Device is DOA

mdframe, 2024-01-24:
Jean, 

I received my new device today, added the power connectors and the E-Paper per instructions.  I DID NOT apply any external power.  Only using 5 different USB-C cords and two different computers I get nothing.  No connection no sign of life.  What can I check on this board for valid power and connection with the USB-C?

Thanks, 

Matt

Brockhold, 2024-01-24:
Hey, I actually thought the same thing happened to me today, [but it turns out I didn't read the manual](https://smart-powermeter.readthedocs.io/en/latest/programming.html) - to show up as a USB serial device for loading firmware, after connecting the USB cable you have to hold down the FLASH button and then press the RESET button.

mdframe, 2024-01-24:
Ben,

Thank you, I've read this document several times and not sure how I missed that!  I will try this one I get home.

Appreciate the help!

Matt

JGAguado, 2024-01-24:
Thanks for answering @Brockhold ! Since it is a recurrent issue, I will emphasize it on the [FAQs](https://smart-powermeter.readthedocs.io/en/latest/faq.html)

I will close now the topic, feel free to reopen it if it didn't work! 

Best regards!

Jon

closed 2024-01-24 by JGAguado