JGAguado/Smart_Powermeter #16: `SCT-013-000` with the built-in 22R resistor does not seem to have enough resolution

roobre, 2024-01-24:
Hi there!

Yesterday I finished assembling my smart power meter but I'm struggling to get it to measure reliably. Trying to go through the calibration step, the device does not seem to be able to discern no current (0A) from 2.1A as measured by an external ammeter (which also matches the expected draw for the load I connected).

The YAML I'm using is as follows:
```
sensor:
  - platform: adc
    pin: GPIO1
    id: Input_1
    attenuation: 11db
    update_interval: 1s
# ...
  - platform: ct_clamp
    name: Probe 1
    id: Probe_1
    sensor: Input_1
    sample_duration: 200ms
    update_interval: 1s
    # filters:
    #   - calibrate_linear:
    #       - 0 -> 0
    #       - 0.042 -> 2.72
# ...
```

Notice the filter section is commented out for calibration. Looking at the logs for lines like:

```
[22:55:16][D][sensor:094]: 'Probe 1': Sending state 0.00176 A with 2 decimals of accuracy
```

I collected ~20 of those lines for a 0A and a 2.1A load and crunched some numbers on a spreadsheet. From that, I got that:
- A 0A load reported an average of `0.001685` with an standard deviation of `0.00028`
- A 2.1A load reported an average of `0.001727` with an standard deviation of `0.00026`

What I understand from these numbers is that measurements of 0W (0A) and ~500W (~2A) are within standard deviation from one another, making them hardly distinguishable.

I wonder if this is expected, which in my opinion would make this meter not very suitable for domestic loads (for which I would hope to have a resolution of 100W, ~0.5A), or if I'm doing something wrong. I'm fine hacking the meter and replacing the 22R resistors with something higher, but I would like to double-check with you to see if what I'm seeing is the expected behavior, or if on the contrary I'm doing something wrong.

[measurements.ods](https://github.com/JGAguado/Smart_Powermeter/files/14037611/measurements.ods)

roobre, 2024-01-24:
Figured this out. The adc was reading noise because, at least on my unit, it doesn't seem like you should plug the jacks all the way into the receptacles. If you do (like I did), then they don't seem to make proper contact. By pulling them out about ~1mm from the point where they would bottom out, I now get sensical readings from the ADC: `0.021 +/- 0.0002` for ~2A, and `0.042 +/- 0.0004` for ~4A.

closed 2024-01-24 by roobre

JGAguado, 2024-01-24:
Hi @roobre, 

Sorry for not having answered before. Good to know that you fixed it! I will add this issue to the [FAQ](https://smart-powermeter.readthedocs.io/en/latest/faq.html)  section on the docs if you don't mind 😃 

All best

Jon

PS: Qué bien ver un usuario español apoyando el proyecto 😄