JGAguado/Smart_Powermeter #17: Failed to Initialize

mdframe, 2024-01-24:
After pressing the flash button and reboot button to get the comm connected, when I try to install a package all I get is "Failed to initialize. Try resetting your device or holding the BOOT button while selecting your serial port until it starts preparing the installation."

I have tried different button sequences and holding buttons down but it always has the same error message.  Funning thing is through all this I can see the USB-C stay connected.

mdframe, 2024-01-24:
Requires multiple pushed between flash and reboot while trying to install.

closed 2024-01-24 by mdframe

Saentist, 2024-02-06:
> Requires multiple pushed between flash and reboot while trying to install.

Can you describe more detailed, I have same problem.